gl-sergei/u2f-token #1: __process6_stack_size__  is not defined in u2f.ld

anvol, 2017-08-20:
Following error received while linking

`u2f-token/u2f/pbt.c:114: undefined reference to '__process6_stack_size__'`

I guess, we need to reserve some space on stack in **u2f.ld**

board.h -> board-blue-pill.h

gl-sergei, 2017-08-20:
You need to put some effort to make it run on blue pill. Since this board doesn't have user buttons, you probably want to revert to commit e12df0f11de14084265d3d403f703abe028a7a4f.

closed 2017-08-20 by gl-sergei

anvol, 2017-08-20:
Yes, thanks! 
FYI - device works as expected even using reset btn as user action (macOS, Chrome).
I think there might be easier to reroute/resolder reset btn to different pin and use it as user btn.

And thanks again for u2f project! Just what I was looking for)

gl-sergei, 2017-08-20:
True. It works because device indicates user presence during the first 10 seconds after startup. I've missed that bit.

commit 1fe9cb15 (2017-08-20): Add missing changes to linker script

Fixes #1