gl-sergei/u2f-token #2: Read Out Protection (RDP) is not set

anvol, 2017-08-23:
The Read Out Protection is a global Flash memory read protection allowing the embedded
firmware code to be protected against copy, reverse engineering, dumping using debug
tools or other means of intrusive attack. This protection should be set by the user after the
binary code is loaded to the embedded Flash memory.

When the read protection level 1 is activated, no access (read, erase, and program) to Flash
memory or backup SRAM can be performed via debug features such as Serial Wire or
JTAG, even while booting from SRAM or system memory bootloader.
However, when booting from Flash memory, accesses to this memory and to backup SRAM
from user code are allowed.
Any read request to the protected Flash memory generates a bus error.
Disabling RDP level 1 protection by re-programming RDP option byte to level 0 leads to a
mass erase.

RDP must be set to Level 1 right after flashing firmware or on the first run.
I think that verify&fix RDP at firmware level is more error-free. We can set OptionBits in the following way at first run (STM32 HAL used)

``` C
void TurnOnRDP() {
    FLASH_OBProgramInitTypeDef obConfig;
 
    HAL_FLASHEx_OBGetConfig(&obConfig);
 
    if (obConfig.RDPLevel == OB_RDP_LEVEL_0) {
        // this is first time we run mcu after flashing firmware
        obConfig.RDPLevel = OB_RDP_LEVEL_1;
        HAL_FLASH_Unlock();
        HAL_FLASH_OB_Unlock();
        HAL_FLASHEx_OBProgram(&obConfig);
        HAL_FLASH_OB_Launch();
    }
}
```

gl-sergei, 2017-08-31:
Agreed. Though it is easy to do with OpenOCD, it should be documented and/or done by firmware on first run.

gl-sergei, 2017-08-31:
Partly fixed (in readme) by 871eb2f506e44f1140241e894dd5fff1f755bad8

gl-sergei, 2018-04-14:
Build with:
```
make TARGET=BLUE_PILL ENFORCE_DEBUG_LOCK=1
```
to enable RDP automatically

PR #7 (2018-04-14): Toboot API 2.0, RDP and more
fixes #2
closed 2018-04-14 by gl-sergei