gl-sergei/u2f-token #9: My led is PB12

assasinfil, 2018-07-30:
How can I reassign LED pb13 to pb12 in the code?

assasinfil, 2018-07-30:
I work witch [stm32f103c6](https://ru.aliexpress.com/item/5pcs-STM32F103C8T6-ARM-STM32-Minimum-System-Development-Board-Module-For-arduino/32653839788.html?spm=a2g0s.9042311.0.0.274233edrvwsF4) and led pin is PB12. I for a long time could not understand why the standard example blink does not work and having looked closely to the board I saw that the LED is connected to the foot of pb12

gl-sergei, 2018-07-30:
For PB12 it should look like

```
#define GPIO_LED_BASE   GPIOB_BASE
#define GPIO_LED_CLEAR_TO_EMIT          12
```

and probably this line needs to be changed to something like
```
#define VAL_GPIO_LED_CRH            0x88838888      /* PC15...PC8 */
```

LED will blink when you plug the board and it will blink each time board communicates via USB.

assasinfil, 2018-07-30:
i changed, but led didn't work((

assasinfil, 2018-07-30:
[image]

gl-sergei, 2018-07-30:
I happened to have this board. Just tested it, it works with these changes:
```diff
diff --git i/board/board-blue-pill.h w/board/board-blue-pill.h
index 68843d7..8fe4fe9 100644
--- i/board/board-blue-pill.h
+++ w/board/board-blue-pill.h
@@ -9,8 +9,8 @@
 #define STM32_PLLMUL_VALUE              9
 #define STM32_HSECLK                    8000000
 
-#define GPIO_LED_BASE   GPIOC_BASE
-#define GPIO_LED_CLEAR_TO_EMIT          13
+#define GPIO_LED_BASE   GPIOB_BASE
+#define GPIO_LED_CLEAR_TO_EMIT          12
 #define GPIO_USB_BASE   GPIOA_BASE
 #define GPIO_USB_SET_TO_ENABLE          12
 #undef  GPIO_OTHER_BASE
@@ -29,8 +29,8 @@
 #define VAL_GPIO_USB_CRH            0x88811888      /* PA15...PA8 */
 
 #define VAL_GPIO_LED_ODR            0xFFFFFFFF
-#define VAL_GPIO_LED_CRL            0x88888888      /*  PC7...PC0 */
-#define VAL_GPIO_LED_CRH            0x88388888      /* PC15...PC8 */
+#define VAL_GPIO_LED_CRL            0x88888888      /*  PB7...PB0 */
+#define VAL_GPIO_LED_CRH            0x88838888      /* PB15...PB8 */
 
-#define RCC_ENR_IOP_EN      (RCC_APB2ENR_IOPAEN | RCC_APB2ENR_IOPCEN)
+#define RCC_ENR_IOP_EN      (RCC_APB2ENR_IOPBEN | RCC_APB2ENR_IOPCEN)
 #define RCC_RSTR_IOP_RST    (RCC_APB2RSTR_IOPARST | RCC_APB2RSTR_IOPCRST)
```

Make sure to run
```
make clean distclean && make TARGET=BLUE_PILL
```

after applying the patch.

Please also note that since your board does not have a push button to check for user presence, firmware will allow register/authenticate only within 10 seconds after device has been plugged in. You'll probably need to unplug the device and plug it back (or press HW reset button) to authenticate.

assasinfil, 2018-07-30:
Nice its work! pls add it to compiler (Target=DARK_PILL)

closed 2018-07-30 by assasinfil

assasinfil, 2018-07-30:
The last question: where do I connect the button for authorization?

gl-sergei, 2018-07-31:
First, you will need to solder a bush button. Then, you will need to modify `pbt.c` to read events on this button's GPIO PIN and probably `board-blue-pill.h` as well, to initialise that GPIO PIN.

For new makefile target I created https://github.com/gl-sergei/u2f-token/issues/10