gl-sergei/u2f-token #14: Able to build & flash blue pill, but no usb device when connected to PC

sklv, 2018-09-17:
Hi, thanks for writing a cool firmware. I'm trying to build and flash a blue pill board. I can build fine (bins.zip containing .elf and .bin attached).
I can't exactly follow your instructions as if I telnet to my localhost, openocd drops the connection, so I opted to use `openocd -f openocd.txt`.
If I use `program build/u2f.bin 0x08000000 verify reset` to flash, when I connect the board to my PC I don't get anything in `dmesg`. If I use `program build/u2f.elf verify reset exit` to flash, I get:

```
** Programming Started
auto erase enabled
Info : device id = 0x20036410
Info : flash size = 64kbytes
Warn : no flash bank found for address 0x07ff0000
Info : Flash write discontinued at 0x08007dac, next section at 0x0800f800
target halted due to breakpoint, current mode: Thread
xPSR: 0x61000000 pc: 0x2000003a msp: 0x20005000
target halted due to breakpoint, current mode: Thread
xPSR: 0x61000000 pc: 0x2000003a msp: 0x20005000
wrote 30720 bytes from file build/u2f.elf in 1.764730s (17.000 KiB/s)
** Programming Finished **
** Verify Started **
Error: timed out while waiting for target halted
target halted due to debug-request, current mode: Handler HardFault
xPSR: 0x81000003 pc: 0x08000150 msp: 0x20004fe0
Error: error executing cortex_m crc algorithm
** Verify Failed **
```

Both BOOT pins set to 0, which has seemed to work fine so thus far.
Any idea what's going wrong? Happy to provide any additional info.

[openocd.txt](https://github.com/gl-sergei/u2f-token/files/2389876/openocd.txt)
[bins.zip](https://github.com/gl-sergei/u2f-token/files/2389865/bins.zip)

gl-sergei, 2018-09-17:
Hi @sklv ,

I've checked your binary and it doesn't work for me either. Are you by chance using Ubuntu 18.04 LTS? Apparently, there is a bug in newlib which causing any binary using newlib to crash. Here is the bug report https://bugs.launchpad.net/ubuntu/+source/gcc-arm-none-eabi/+bug/1767223 and one possible workaround is to install newer package https://github.com/bbcmicrobit/micropython/issues/514#issuecomment-404759614

assasinfil, 2018-09-17:
Use kali linux, its best os)))

sklv, 2018-09-17:
I tried with the unlock commands enabled, and all configurations of the bootloader pins to get the same results as before.
I'm using gentoo with a toolchain built with `crossdev -s4 -S --ex-gdb -t arm-none-eabi` which i have been led to believe in the openocd irc, might be the cause of my issues.
I will try to non-self-built toolchain and see if it makes a difference.

gl-sergei, 2018-09-17:
I've seen [this thread](https://lists.gnupg.org/pipermail/gnuk-users/2018-August/000077.html) with similar issues on Gentoo caused by incorrect build flags.

sklv, 2018-09-18:
I've verified that my toolchain is the issue, as I compiled with the toolchain from arm.com and everything is working.

closed 2018-09-18 by sklv