gl-sergei/u2f-token #18: undefined_symbol: hid_enumerate

matmunn, 2019-02-08:
Trying to run `./certtool init` is throwing the following errors:
```
Traceback (most recent call last):
  File "./certtool", line 139, in <module>
    main()
  File "./certtool", line 136, in main
    args.func(args)
  File "./certtool", line 102, in command_init
    e = easyhid.Enumeration()
  File "/home/server/Dev/venv/lib/python3.7/site-packages/easyhid/easyhid.py", line 358, in __init__
    self.device_list = _hid_enumerate(vid, pid)
  File "/home/server/Dev/venv/lib/python3.7/site-packages/easyhid/easyhid.py", line 420, in _hid_enumerate
    start = hidapi.hid_enumerate(vendor_id, product_id)
  File "/home/server/Dev/venv/lib/python3.7/site-packages/cffi/api.py", line 882, in __getattr__
    make_accessor(name)
  File "/home/server/Dev/venv/lib/python3.7/site-packages/cffi/api.py", line 878, in make_accessor
    accessors[name](name)
  File "/home/server/Dev/venv/lib/python3.7/site-packages/cffi/api.py", line 808, in accessor_function
    value = backendlib.load_function(BType, name)
AttributeError: function/symbol 'hid_enumerate' not found in library '<None>': python: undefined symbol: hid_enumerate
(venv) [server@server cert]$ sudo ./certtool init
Traceback (most recent call last):
  File "./certtool", line 139, in <module>
    main()
  File "./certtool", line 136, in main
    args.func(args)
  File "./certtool", line 102, in command_init
    e = easyhid.Enumeration()
  File "/home/server/Dev/venv/lib/python3.7/site-packages/easyhid/easyhid.py", line 358, in __init__
    self.device_list = _hid_enumerate(vid, pid)
  File "/home/server/Dev/venv/lib/python3.7/site-packages/easyhid/easyhid.py", line 420, in _hid_enumerate
    start = hidapi.hid_enumerate(vendor_id, product_id)
  File "/home/server/Dev/venv/lib/python3.7/site-packages/cffi/api.py", line 882, in __getattr__
    make_accessor(name)
  File "/home/server/Dev/venv/lib/python3.7/site-packages/cffi/api.py", line 878, in make_accessor
    accessors[name](name)
  File "/home/server/Dev/venv/lib/python3.7/site-packages/cffi/api.py", line 808, in accessor_function
    value = backendlib.load_function(BType, name)
AttributeError: function/symbol 'hid_enumerate' not found in library '<None>': python: undefined symbol: hid_enumerate
```

I've tried on both macOS 10.12.6 and a Linux system running kernel 4.20.6. As far as I can tell I've installed all the dependencies required, I've also tried Python 3.6.8 and 3.7.2

gl-sergei, 2019-02-08:
It looks like `easy_hid` package cannot find hidapi library. This library is available as brew formula on OS X and as a package on debian/ubuntu.

matmunn, 2019-02-08:
Yep, that was the problem. Seems to be all good now, thanks for that

closed 2019-02-08 by matmunn

V3ntus, 2022-10-21:
I'm having this issue as well, although I've followed instructions with installing `hidapi` with brew. I've unlinked and relinked it as well. Using Python 3.10. Not sure where to start with troubleshooting