gl-sergei/u2f-token #21: Different udev rules

asymmetric, 2019-02-09:
The `udev` rules in the README are different from the ones on the tomu.im homepage:

README:

```
ACTION=="add|change", KERNEL=="hidraw*", SUBSYSTEM=="hidraw", ATTRS{idVendor}=="16d0", ATTRS{idProduct}=="0e90", TAG+="uaccess"
```

tomu.im:

```
ACTION=="add|change", KERNEL=="hidraw*", SUBSYSTEM=="hidraw", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="cdab", TAG+="uaccess"
```

Is this intended? And how does it affect the functioning of the device?

gl-sergei, 2019-02-09:
The firmware in this repo has 16d0:0e90 VID:PID. TAG+="uaccess" rule allows non privileged users to access hid device associated with given VID:PID.

asymmetric, 2019-02-09:
I see. So do you reckon this means the tomu.im instructions (they mention this repo) are wrong?

gl-sergei, 2019-02-09:
This project started as u2f implementation for stm32f01. Then I made efm32hg port. It is now hosted here: https://github.com/im-tomu/chopstx. Since then, I decided that it will be easier to maintain both version in the single source tree. The firmware in im-tomu repo (as this one originally used to) is using VID 0483. However that VID belongs to STMicroelectronics and its usage is illegal. I opted out for obtaining a PID from MCS Electronics for this project recently (https://www.mcselec.com/index.php?page=shop.product_details&flypage=shop.flypage&product_id=92&option=com_phpshop&Itemid=1).

closed 2019-02-09 by asymmetric