gl-sergei/u2f-token #25: Add .elf file to releases, or make inject_key work on .bin files

eliotb, 2019-03-20:
Currently Releases only contains .bin files, but in order to use inject_key.py,  the .elf file is needed.

So, right now can only use inject_key if binary is built from source..

gl-sergei, 2019-03-20:
You still need to install arm toolchain if you want to use inject_key. So, having downloadable .elf only will save you one command :)

eliotb, 2019-03-20:
Ah ok.  I built my own bin, but for some reason it seemed not to work, while your one does.  However I have no way to inject my key into your .bin ...

gl-sergei, 2019-03-20:
Which OS/Linux distro do you use? There are known issues with Ubuntu 18.04 (https://bugs.launchpad.net/ubuntu/+source/gcc-arm-none-eabi/+bug/1767223) and I've heard of similar issues on Gentoo as well. I think downloading and using it for build may help https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads

eliotb, 2019-03-21:
OS is Debian 9.  I downloaded the arm toolchain you suggested, and rebuilt.   New tomu works well - thanks. I can't definitely say the new tool is the reason for the improvement (I haven't proved why the original didn't work correctly).
For reference here is how I invoked the specific downloaded tool:
`make CROSS=/home/eliot/installed/gcc-arm-none-eabi-8-2018-q4-major/bin/arm-none-eabi- TARGET=TOMU ENFORCE_DEBUG_LOCK=1`

Still as I think you mentioned on another thread, it would be nice to be able to patch a saved key into the binary directly without needing the ARM tools. 
Also to patch a new auth counter into a binary, useful for making a backup device along the lines of this https://dmitryfrank.com/articles/backup_u2f_token

gl-sergei, 2019-03-21:
Agree. Lets keep this issue as a request to change inject_key.py to work on ".bin" files instead of ".elf" ones.

eliotb, 2019-03-21:
I will have proof-of-concept pull request soon.  (for TOMU only at first)

eliotb, 2019-03-22:
I'm having some trouble with inject keys.  I can use your release .bin file, but after patching it doesn't work right.
Looking at .ld file and map file, 
This contains attestation_cert@f400, device_key@f800 and auth_ctr@fc00.   
inject_key overwrites whole .flash_storage, but I wonder should attestation cert be overwritten? Or prepended to the blob that is written to that section?
I'll change my inject_key_bin.py to modify at addresses as above and see if that works...

eliotb, 2019-03-23:
BTW I think the toolchain change was not necessary.  The failure I think came from building using git head, and failing to do "certtool init" after programming the key.

gl-sergei, 2019-03-23:
Yeah. I changed default flag to skip injection of attestation key without updating the readme.
Personally, I am not a big fan of having a backup of your u2f key. You can have spare device as a backup. And if you lost your device, you should immediately unlink it from all your accounts. Because the u2f device by design is not protected from being used by someone who maybe already knows your password.

eliotb, 2020-03-26:
I see my modified inject_key.py has been merged.
Also, I now understand that you can register two devices with different secret keys with a service e.g. github, keeping one somewhere secure for backup.

closed 2020-03-26 by eliotb