gl-sergei/u2f-token #29: Errors while using u2f-token with tomu

AlexKapustin, 2019-05-02:
Hello!

I'm trying to use my Tomu (v.0.4) as an U2F-Token
There are few cases which produce different errors:

**Case #1**
I compile firmware by command:
`make TARGET=TOMU ENFORCE_DEBUG_LOCK=0`
and than flash it by command:
`sudo dfu-util -v -d 1209:70b1 -D build/u2f.bin`
BTW: phrase from readme.md
> Binaries for Tomu are built with bootloader support, use the following command to flash the firmware:
> dfu-util -d 1209:70b1 -D u2f-TOMU.bin

is confusing.

Then I use https://github.com/mdp/u2fcli to test my token.
When I insert dongle I can see in logs:
```
[ 2483.341307] usb 1-4: USB disconnect, device number 14
[ 2487.385776] usb 1-4: new full-speed USB device number 15 using xhci_hcd
[ 2487.536935] usb 1-4: New USB device found, idVendor=16d0, idProduct=0e90
[ 2487.536941] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2487.536946] usb 1-4: Product: U2F-token (EFM32)
[ 2487.536950] usb 1-4: Manufacturer: unknown
[ 2487.536953] usb 1-4: SerialNumber: 1.00
[ 2487.540344] hid-generic 0003:16D0:0E90.000A: hiddev0,hidraw0: USB HID v1.10 Device [unknown U2F-token (EFM32)] on usb-0000:00:14.0-4/input0
```

when I run:
`./u2fcli reg --challenge complexChallengeGoesHere12345671 --appid https://mdp.im`
I have next response:
```Registering, press the button on your U2F device #1 [unknown U2F-token (EFM32)]Error registering with device: u2ftoken: unexpected error 27014 during registration```

Dongle does not react on command at all.
After some investigations I can see in `src/cert/certificates.c`
```
struct attestation_cert  __attribute__ ((section(".attestation.cert")))
attestation_cert = {
  .der_len = (uint32_t) -1,
  .der = NULL,
  .key = NULL
};
```
I guess it's not expected

**Case #2** 
I compile firmware with next command:
` make TARGET=TOMU ENFORCE_DEBUG_LOCK=0`
then I generate certificate and inject it to firmware:
```
openssl ecparam -name prime256v1 -genkey -noout -outform der -out key.der
./inject_key.py --key key.der
```
then, I flash it .

When insert in logs:
```[ 3398.004924] usb 1-4: new full-speed USB device number 19 using xhci_hcd
[ 3398.155269] usb 1-4: New USB device found, idVendor=16d0, idProduct=0e90
[ 3398.155275] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 3398.155280] usb 1-4: Product: U2F-token (EFM32)
[ 3398.155284] usb 1-4: Manufacturer: unknown
[ 3398.155287] usb 1-4: SerialNumber: 1.00
[ 3398.158837] hid-generic 0003:16D0:0E90.000C: hiddev0,hidraw0: USB HID v1.10 Device [unknown U2F-token (EFM32)] on usb-0000:00:14.0-4/input0
```

Testing:
```
./u2fcli reg --challenge complexChallengeGoesHere12345671 --appid https://mdp.im

lengeGoesHere12345671 --appid https://mdp.im
Registering, press the button on your U2F device #1 [unknown U2F-token (EFM32)]
```
Dongle starts to blink by red led with impulse pattern. When I touch it changes blink by solid red light. In console I can see:
```Error registering with device: u2fhid: error reading response, read timed out```
```

I will continue to investigate why there is NULL in certificates.c file and update you as soon as I find something.

Thank you in advance!

Best regards,
Alex

AlexKapustin, 2019-05-02:
After I compiled:
```make TARGET=TOMU ENFORCE_DEBUG_LOCK=0 CUSTOM_ATTESTATION_CERT=1
```
it seems work ok now.
This parameter is a bit confusing. (Maybe we should update readme.md file ?)

```
ifeq ($(CUSTOM_ATTESTATION_CERT),1)
        GENCERT_CMD = cd cert && ./gen.sh && ( python dump-der.py > certificates.c || ( rm certificates.c && exit 1 ) )  && cd ..
else
        GENCERT_CMD = cp empty-attestation-cert.c cert/certificates.c
endif
```

gl-sergei, 2019-05-02:
The very first of the binaries you produced was OK. The "initialize device" step is necessary after flashing with dfu-util (https://github.com/gl-sergei/u2f-token#initialize-device). Instead of building with CUSTOM_ATTESTATION_CERT=1, you supposed to use the one which comes pre-generated. And you could use `certtool` to upload it.

gl-sergei, 2019-05-02:
So, in general, the whole procedure of building and flashing should look like:
make TARGET=TOMU (or download the binary from the releases page)
dfu-util -D
./certtool init

in this case you avoid embedding any secrets in your binary which I am all for.

Other options like injecting the custom secret key or certificate into the binary are for those who feel adventurous and can dig into the source code like you did .

If you feel that readme needs to be improved and know how exactly, pull requests are welcome :)

AlexKapustin, 2019-05-03:
Thank you for your reply!
I will prepare MR next week.

gl-sergei, 2019-05-07:
Fixed by https://github.com/gl-sergei/u2f-token/pull/31

closed 2019-05-07 by gl-sergei