gl-sergei/u2f-token #33: Alternative to NIST P-256 ?

Debian-10, 2019-05-11:
I try to figure out how to get the token with something else than NIST P-256, which can be considered as unsafe. I already tried with ed25519 , but it blocks with inject_key.py .
Does someone have an idea ?

gl-sergei, 2019-05-11:
What do you want ed25519 for? U2F spec and all implementations require p-256 for signature. If you implement other curve/signing algorithm, you won't be able to use it with any existing u2f implementation.

Debian-10, 2019-05-11:
Thank you for answering me fast,
I meant for more safety (suspect a backdoor on nist p-256), but if you say it's a standard , i'll keep it.

gl-sergei, 2019-05-11:
U2F standard does not give an option to choose the signature algorithm. Newer FIDO2 allows 2048-bit RSA with SHA-256 in addition to ECDSA over P-256 with SHA-256. But FIDO2 is out of scope of this project.

closed 2019-05-11 by gl-sergei