gl-sergei/u2f-token #38: Unable to boot properly with TOMU inserted

mrvanes, 2019-06-16:
With my tomu (using u2f-token fw) inserted into a USB port I'm unable to properly boot my Linux laptop untill I remove the token. The error logged in dmesg is:
```
[   16.923942] usb 1-7: device descriptor read/64, error -110
```
Linux version: (K)Ubuntu 19.04 with custom compiled kernel 5.1.10. I might miss some specific USB module compiled in, but once booted the token works flawless nonetheless:
```
[  646.417952] usb 1-7: new full-speed USB device number 6 using xhci_hcd
[  646.576314] hid-generic 0003:16D0:0E90.0003: hiddev96,hidraw0: USB HID v1.10 Device [unknown U2F-token (EFM32)] on usb-0000:00:14.0-7/input0
```

gl-sergei, 2019-06-17:
Hmm... I am afraid I cannot reproduce this issue. I have been using various kernel versions staring with 4.4 and now I am using 5.1.8 on my arch laptop.

By "cannot properly boot" do you mean that boot process hangs, or that device isn't working after boot?

mrvanes, 2019-06-17:
Yes, the laptop keeps repeating the error until I remove the token and only continues booting when I remove it. After that, the token simply works as expected. I have two USB 3.0 ports, so I can't test for 2.0 vs. 3.0 behaviour.

gl-sergei, 2019-06-17:
Do you the same behaviour with Ubuntu stock kernel? Do other USB devices work flawlessly?

mrvanes, 2019-06-17:
Hmmm... de stock kernel (5.0.0.16) does report the same errors, but silently continues.... weird?

gl-sergei, 2019-06-18:
I think stock kernel's behaviour is more correct.

I do not see any errors in my boot log:
```
[    2.852023] random: fast init done
[    2.860475] usb 1-2: New USB device found, idVendor=16d0, idProduct=0e90, bcdDevice= 1.00
[    2.860480] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    2.860483] usb 1-2: Product: U2F-token (EFM32)
[    2.860485] usb 1-2: Manufacturer: unknown
[    2.860488] usb 1-2: SerialNumber: 1.00
[    2.870246] hidraw: raw HID events driver (C) Jiri Kosina
[    2.872909] usbcore: registered new interface driver usbhid
[    2.872909] usbhid: USB HID core driver
```
then kernel enumerates other devices and then
```
[    4.092640] hid-generic 0003:16D0:0E90.0001: hiddev0,hidraw0: USB HID v1.10 Device [unknown U2F-token (EFM32)] on usb-0000:00:14.0-2/input0
```

mrvanes, 2019-06-18:
I agree the stock behaviour is more correct, I'm just wondering what feature does the trick. But that's out of scope of this issue.

closed 2019-06-18 by mrvanes