gl-sergei/u2f-token #39: Windows + Firefox support

mkroman, 2019-10-01:
When trying to use my Tomu device flashed with the current [HEAD](https://github.com/gl-sergei/u2f-token/tree/0fcf93dc232f6724f70baaa68355a2e04a199671) on Windows 10 (Build 18362.356), Firefox 69.0.1 the device is recognized but the communication immediately fails.

[image]

I see the [Initialize device](https://github.com/gl-sergei/u2f-token/tree/0fcf93dc232f6724f70baaa68355a2e04a199671#initialize-device) section - does this also apply to locally built firmware?

gl-sergei, 2019-10-01:
Yes, it does. Initialization uploads attestation certificate on device. As an option you can build attestation certificate into firmware itself using following commands:

```
cd cert
python dump-der.py > certificates.c
```

and then clean build the firmware (`make clean && make distclean && make`)

mkroman, 2019-10-13:
I got the time to try building and flashing my Tomu again, doing a `make distclean && make dist clean` beforehand, then doing the
```
cd cert
python dump-der.py > certificates.c
```
commands - followed by `make TARGET=TOMU ENFORCE_DEBUG_LOCK=1`.

To flash the device I did a `sudo dfu-util -v -d 1209:70b1 -D build/u2f.bin` but no dfu-capable device would show up. I tried unplugging it and plugging it again to no avail. Then I ran `while true; do sudo dfu-util -v -d 1209:70b1 -D build/u2f.bin; done` while slowly inserting the Tomu, but I didn't see it successfully uploading anything and I gave up.

I'm now back in Windows and I decided to test it out - both the YubiKey test page and GitHub no longer shows the "Can't read your security key." message and registering the key seems to work just fine.

I'm now using Firefox 69.0.3 and Windows 10 (Build 18362.418).

This leaves me with the conclusion that either
1. It managed to upload the new firmware and I missed it
2. Firefox or Windows fixed some compatibility issue
3. Maybe the u2f-token firmware somehow still hadn't initialized the attestation certificate (as I understand it this is the default behavior if none is provided?)

Number 1 is probably the most likely scenario. In any case, thank you for your reply, I'll close this issue.

mkroman, 2019-10-13:
This is the attestation certificate the YubiKey demo page gave me:

```
Version: 3 (0x2)
Serial Number: 10623373676140538000 (0x936dce12ca97a800)
Signature Algorithm: ecdsa-with-sha256
Issuer: CN = U2F Token
Validity
Not Before: 2019-03-10T12:06:56
Not After: 2029-03-07T12:06:56
Subject: C=US,CN=U2F Token
Subject Public Key Info:
Public-Key: (256 bit)
pub:

04:26:02:ac:c4:34:33:7d:79:58:85:97:9b:ae:24:1b:10:58:
a4:4e:7e:b1:6d:00:de:dc:ff:04:9f:cf:00:18:ad:0a:8c:96:
ac:12:28:f7:53:06:ad:fe:75:c6:d5:7c:44:d3:c2:30:5d:65:
1f:e5:21:d0:2f:7e:a1:c7:24:8f:ff

Curve: secp256r1
X509v3 extensions:
1.3.6.1.4.1.45724.2.1.1 (fidoU2FTransports):
Hex value: 03020520
Transports: USB
Signature Algorithm: ecdsa-with-sha256

30:45:02:20:74:22:f7:02:82:66:c3:4d:26:2e:1c:de:28:0d:
b7:36:d2:9c:30:e1:4c:81:60:e7:a3:bb:78:64:40:ee:be:3d:
02:21:00:91:4b:bf:6f:6c:59:8c:a9:cd:d2:38:5b:23:dd:3a:
98:92:51:69:07:dc:99:e2:d3:c1:64:01:d2:4b:65:99:13
```

or PEM:

```
-----BEGIN CERTIFICATE-----
MIIBTzCB9qADAgECAgkAk23OEsqXqkowCgYIKoZIzj0EAwIwITELMAkGA1UEBhMC
VVMxEjAQBgNVBAMMCVUyRiBUb2tlbjAeFw0xOTAzMTAxMjA2NTZaFw0yOTAzMDcx
MjA2NTZaMCExCzAJBgNVBAYTAlVTMRIwEAYDVQQDDAlVMkYgVG9rZW4wWTATBgcq
hkjOPQIBBggqhkjOPQMBBwNCAAQmAqzENDN9eViFl5uuJBsQWKROfrFtAN7c/wSf
zwAYrQqMlqwSKPdTBq3+dcbVfETTwjBdZR/lIdAvfqHHJI//oxcwFTATBgsrBgEE
AYLlHAIBAQQEAwIFIDAKBggqhkjOPQQDAgNIADBFAiB0IvcCgmbDTSYuHN4oDbc2
0pww4UyBYOeju3hkQO6+PQIhAJFLv29sWYypzdI4WyPdOpiSUWkH3Jni08FkAdJL
ZZkT
-----END CERTIFICATE-----
```

The issue date leads me to believe that it may have been in an unready state for quite a while when I tried to test it - is this possible?

closed 2019-10-13 by mkroman