gl-sergei/u2f-token #40: The key is not initialized

assasinfil, 2019-11-13:
certtool:
```
Trying to initialize device HIDDevice:
    /dev/hidraw1 | 16d0:e90 | unknown | U2F-token (STM32) | 1.00
    release_number: 256
    usage_page: 14898
    usage: 11825
    interface_number: 0
APDU Error: 6986
```
dmesg:
```
[  184.219613] usb 1-2: new full-speed USB device number 6 using xhci_hcd
[  184.368774] usb 1-2: New USB device found, idVendor=16d0, idProduct=0e90, bcdDevice= 1.00
[  184.368778] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  184.368781] usb 1-2: Product: U2F-token (STM32)
[  184.368783] usb 1-2: Manufacturer: unknown
[  184.368785] usb 1-2: SerialNumber: 1.00
[  184.372322] hid-generic 0003:16D0:0E90.0003: hiddev0,hidraw1: USB HID v1.10 Device [unknown U2F-token (STM32)] on usb-0000:00:14.0-2/input0
```

bolshevik, 2020-03-15:
In my case recompiling without `ENFORCE_DEBUG_LOCK=1` and erasing the whole flash with `stm32f1x mass_erase 0` helped. But after the initialization it makes sense to lock the flash with `stm32f1x lock 0`.

legioner0, 2020-03-30:
#45 - one more case for that problem

yashikada, 2023-05-21:
I think, you can use the same solution which I described [here](https://github.com/gl-sergei/u2f-token/issues/45#issuecomment-1556121016)

closed 2024-02-18 by assasinfil