gl-sergei/u2f-token #41: blue pill switch?

Qwertylex, 2020-01-20:
I flashed the prebuilt to my blue pill and it seems reset button works as the auth button
and it cycles usb disconnect/connect in kernel messages, and it works on demo.yubico.com as you'd expect it should normally work...

is the reset button really intended to be the auth button or is this unexpected?

otherwise, where should a button go?

Qwertylex, 2020-01-21:
I guess I don't understand how [uvoid.c](https://github.com/gl-sergei/u2f-token/blob/master/src/uvoid.c) works, sorry I'm a bit of a noob. I would like to hear how it all works though.

Also I have [added support](https://github.com/Qwertylex/u2f-token) for using a switch on PA10 on the blue pill rather than using the reset button.

retroban, 2020-08-19:
PA10 --- >>> GND or PA10 --- >>> + 3.3V? How to connect correctly and what function does the button perform?

HinataKato, 2020-12-04:
> I flashed the prebuilt to my blue pill and it seems reset button works as the auth button
> and it cycles usb disconnect/connect in kernel messages, and it works on demo.yubico.com as you'd expect it [...]

I think it is the expected behavior, let the bluepill as a non-button key, reset button is equivalent to re-insert the security key