gl-sergei/u2f-token #45: Unable to intialize device with injected key

legioner0, 2020-03-30:
I've tried to use u2f with injected key:
```
make TARGET=ST_DONGLE
./inject_key.py --key key.der
openocd -f interface/stlink-v2.cfg -f target/stm32f1x.cfg
```
on step "Initialize device" I've got:
```
python3 certtool init
Trying to initialize device HIDDevice:
    /dev/hidraw4 | 16d0:e90 | unknown | U2F-token (STM32) | 1.00
    release_number: 256
    usage_page: 14898
    usage: 11825
    interface_number: 0
APDU Error: 6986
```

After some investigation I found that the problem with attestation certificate - `empty-attestation-cert.c` conflicts with injected key.
Workaround: `rm -f cert/certificates.c && make TARGET=... CUSTOM_ATTESTATION_CERT=1`

yashikada, 2023-05-21:
I think the issue is in the script inject_key.py which replace section flash_storage and set devicekey at position 0x0 (0x0F400 in flash) while the correct position is 0x400 (0x0F800 in flash).
When devicekey is set the firmware think the attestation key is already initialized and return error.
I used the script inject_key_bin.py (`python3 inject_key_bin.py --key key.der --ctr 100 --bin build/u2f.bin`) which don't replace section flash_storage but patch the exaction position, so the attestation cert remain 0xff (-1) and when certtool init is invoked attestation cert is set.

gl-sergei, 2023-05-21:
@yashikada thank you. Indeed, `inject_key.py` produced broken flash layout. https://github.com/gl-sergei/u2f-token/pull/50 should fix it. Unfortunately, I don't have any devices around to test it. Can you please check if it works OK?

yashikada, 2023-05-25:
Sorry for the delay, yes the PR works.
Thank you

yashikada, 2023-05-25:
@gl-sergei sorry wrong test, doesn't works. I tested on demo site, report timeout, I tested on google report wrong key.

gl-sergei, 2023-05-27:
@yashikada  thank you for your test. I  was  finally able to  set up stand and verify and fix it on stm32. Now it works for me. I'll close this issue. Feel free to reopen it if there are still any problems.

closed 2023-05-27 by gl-sergei

yashikada, 2023-05-28:
for me is working, thank you for the improvement.

commit 158e4d62 (2023-05-21): Fix #45: Unable to initialize device with injected key

1. Change inject_key.py to preserve this flash storage layout:
   - 1KiB attestation certificate and key
   - 1KiB device key
   - 1KiB auth counter

2. Change cert/gen.sh and friends to produce the same flash layout
   as above
commit b0c0f745 (2023-05-21): Fix #45: Unable to initialize device with injected key

1. Change inject_key.py to preserve this flash storage layout:
   - 1KiB attestation certificate and key
   - 1KiB device key
   - 1KiB auth counter

2. Change cert/gen.sh and friends to produce the same flash layout
   as above
commit 4bd6422e (2023-05-21): Fix #45: Unable to initialize device with injected key

1. Change inject_key.py to preserve this flash storage layout:
   - 1KiB attestation certificate and key
   - 1KiB device key
   - 1KiB auth counter

2. Change cert/gen.sh and friends to produce the same flash layout
   as above
commit 17bf1ab5 (2023-05-21): Fix #45: Unable to initialize device with injected key

1. Change inject_key.py to preserve this flash storage layout:
   - 1KiB attestation certificate and key
   - 1KiB device key
   - 1KiB auth counter

2. Change cert/gen.sh and friends to produce the same flash layout
   as above